paulvha/ThingMagic #1: ThingMagic/use_serialport/

romainrobotics, 2019-03-26:
There doesn't seem to be a file here. Are you able to use MercuryApi connecting the reader to the UART of the Raspberry Pi

paulvha, 2019-03-26:
I am out on vacation on the moment. Will respond next week..

Outlook voor Android downloaden<https://aka.ms/ghei36>

paulvha, 2019-03-27:
Just reading in more detail. The standard mercuri api works on a raspberry pi. Found some later version caused a compiler error and needed to make a number of changes in the source code.
Regards
Paul

Outlook voor Android downloaden<https://aka.ms/ghei36>

romainrobotics, 2019-03-27:
Thanks for looking into it. For now, I'm just having the Pi communicate with an Arduino via UART, then the Arduino controls the Sparkfun SRTR separately.

closed 2021-05-20 by paulvha