paulvha/ThingMagic #3: A few questions

zymurgy0, 2020-09-20:
Hello Paulvha,
Can you tell me which model of Raspberry Pi you used for this and also what type of connection between the sensor and the raspberry pi you used?
Thank you.

paulvha, 2020-09-20:
HI
I connected with USB/Serial breakout (https://www.sparkfun.com/products/14050). I have tested on Jessie release RPI 2 / 3B+
I have recently bought an RPI4, running Pi-OS/Buster but have not tried to compile all the example and code on the Raspberry PI. Those I did try, compiled without issues (after installing the MercuryLibrary and Readline)

zymurgy0, 2020-09-20:
Hello,
With this sparkhub chip, that’s the sensor that detects the rfid? We were
going to try to use a thingmagic m6e-nano.

paulvha, 2020-09-20:
I use the M6E Nano -> Serial/USB -> USB connection on the RaspberryPi. Then you need to download the latest MercuryAPI from Jadak. When you first compile a new version it will show some errors in the code that need to be corrected.

zymurgy0, 2020-09-20:
Ohhh ok. How do you connect the m6e nano to the serial usb?
Thank you.

paulvha, 2020-09-20:
On the M6E-Nano there is connector "Serial". I have put pins on that. I connect the USB/Serial breakout (
> https://www.sparkfun.com/products/14050). to those pins and a cable from the breakout out to one of the USB ports.

Maybe I do not understand your question however...

regards,
Paul

zymurgy0, 2020-09-20:
No you're understanding it. That mostly answers it. I've been looking for
some schematics that indicate what should be soldered where.

paulvha, 2020-09-20:
Drop me a mail directly to paulvha@hotmail.com. I will sent you some pictures.

regards,
Paul

closed 2021-05-20 by paulvha