paulvha/ThingMagic #5: Using external microcontroller UART

rfidreader, 2021-03-30:
I'm fairly new to TM RFID Modules and to controlling anything more than a servo using UART. Where should I start if I'm trying to use a TI MSP430FR2355 to program an M6E Micro? I'm experienced in programming the MSP however I just need some direction on controlling and reading from the module via UART. Any help is appreciated

paulvha, 2021-03-31:
hi,
I am afraid I can not be of much help. I took some time to read about MSP430FR2355 and the development environment. From what I have seen, the board and chip look interesting but the development environment seems very propriety to me. It looks like normal "c", but the coding directly to the different registers and ports.. gives me more the impression it is assembler level language. 

Not sure what your learning plans are in the future, but with other environments the code on at least one level up from Assembler (often multiple levels up) where you don't care about the different register's value and ports. You have API-libraries that provide you the functionality of (e.g. Serial.begin(115200)  which is translated by the API-libraries for the different boards to the right actions on registers)

Honestly.. I would advice to start with Arduino IDE.. which supports many many different boards, has many out of the box libraries and examples. The code is C++ ( close to C) and has different libraries ( e.g WIFI, Bluetooth, sensors etc etc)  as well  for the M6E Nano.

regards,
Paul

rfidreader, 2021-03-31:
Thanks for responding so quickly and taking the time to help! 

I have experience with Arduinos so that was my first choice. However, this project is for my Capstone Class at Uni and my professor told us we had to use the MSP because we've had previous experience with it in another class.

I've reviewed the Mercury API and tried to think of ways to implement it through the MSP but I think you're right. Just to be clear you're essentially saying that I need Object Oriented Programming ability? If I have to go back to said professor and tell him the MSP won't work I'll need a good reason as he will not be happy.

Thanks again Paul.

paulvha, 2021-04-01:
To learn microprocessor (interaction), the closer to assembler the better. I started off many years ago with coding binary on a 8086  and Z80 and could read the binary code of an 80286 processor, knowing what instructions the processor got. I have build my own hardware memory and in the end even wrote a floppy disk formatter in assembler for a TRS80.(took weeks !!)  That way I have learned a lot about the interaction and still gain advantage about that knowledge. That could be the goal of your study, But writing a complete protocol that way ???

The Jadak mercury API is way too big to even consider. Sparkfun made an implementation which is a cut-down version written in C++ for the Arduino. I have enhanced that and posted in https://github.com/paulvha/ThingMagic/tree/master/Arduino_lib_special. One could connect the M6E to an Arduino, turn on debug in the sketch and see what binary code is sent and expected to be received. You can then embulate that same sequence of binary numbers from assembler / C level. You may find already parts of that binary code exchange in posts on https://forum.sparkfun.com/viewforum.php?f=118

regads,
Paul

closed 2021-05-20 by paulvha