paulvha/ThingMagic #6: Sparkfun Artemis compatability?

dyones7, 2021-04-26:
Hello paulvha,

I was looking for a solution to reading a RFID chip with user that extends past the usual 64 bytes and came across you library.

I am running on the Arduino platform but can't see to get you library to compile, for any Arduino.

Do you support the Arduino and if so can you tell me how to set it up so I can try your library.

Most appreciated,

dyones7

paulvha, 2021-04-26:
Hi

This should compile on an Arduino MEGA2560 without any issues. On the Arduino UNO the small memory size might be a problem. What errors do you get so I can see if I can help.

Regards,
Paul

Verzonden vanuit Mail<https://go.microsoft.com/fwlink/?LinkId=550986> voor Windows 10

dyones7, 2021-04-26:
Hello paulvha,

I have attached my modified code and the error for the mega and the artemins. Note I am using Serial1 for both.

I don’t have a lot of experience with the inner working of Arduino. I  brought in your library as a zip file. Since I already have the SparkFun reader library installed, I include you header file by giving it the entire path. Don’t know fi this is a good thing to do or not.

Thanks for you help,
Dale

Arduino: 1.8.13 (Windows 10), Board: "Arduino Mega or Mega 2560, ATmega2560 (Mega 2560)"

C:\Users\Dale\AppData\Local\Temp\ccBUX1lC.ltrans0.ltrans.o: In function `global constructors keyed to 65535_0_Example15_readallbanks_my_changes.ino.cpp.o.1890':

<artificial>:(.text.startup+0xa8): undefined reference to `RFID::RFID()'

C:\Users\Dale\AppData\Local\Temp\ccBUX1lC.ltrans0.ltrans.o: In function `setupNano':

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:289: undefined reference to `RFID::begin(Stream&)'

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:299: undefined reference to `RFID::getVersion()'

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:304: undefined reference to `RFID::stopReading()'

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:321: undefined reference to `RFID::getVersion()'

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:315: undefined reference to `RFID::setBaud(long)'

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:325: undefined reference to `RFID::setTagProtocol(unsigned char)'

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:327: undefined reference to `RFID::setAntennaPort()'

C:\Users\Dale\AppData\Local\Temp\ccBUX1lC.ltrans0.ltrans.o: In function `setup':

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:129: undefined reference to `RFID::setRegion(unsigned char)'

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:131: undefined reference to `RFID::setReadPower(int)'

C:\Users\Dale\AppData\Local\Temp\ccBUX1lC.ltrans0.ltrans.o: In function `loop':

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:140: undefined reference to `RFID::getTemp()'

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:155: undefined reference to `RFID::ReadingAllBanks(TMR_TagReadData*)'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board Arduino Mega or Mega 2560.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Arduino: 1.8.13 (Windows 10), Board: "SparkFun RedBoard Artemis, 921600, SparkFun Variable Loader (Recommended)"

c:/users/dale/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: sketch\Example15_readallbanks_my_changes.ino.cpp.o: in function `_GLOBAL__sub_I_nano':

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:103: undefined reference to `RFID::RFID()'

c:/users/dale/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: sketch\Example15_readallbanks_my_changes.ino.cpp.o: in function `loop':

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:140: undefined reference to `RFID::getTemp()'

c:/users/dale/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:155: undefined reference to `RFID::ReadingAllBanks(TMR_TagReadData*)'

c:/users/dale/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: sketch\Example15_readallbanks_my_changes.ino.cpp.o: in function `setupNano(long)':

C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:289: undefined reference to `RFID::begin(Stream&)'

c:/users/dale/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:299: undefined reference to `RFID::getVersion()'
[... 1 lines cut ...]
c:/users/dale/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:304: undefined reference to `RFID::stopReading()'

[... cut ...]
c:/users/dale/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: C:\Users\Dale\Documents\Arduino\Example15_readallbanks_my_changes/Example15_readallbanks_my_changes.ino:131: undefined reference to `RFID::setReadPower(short)'

collect2.exe: error: ld returned 1 exit status

exit status 1

Error compiling for board SparkFun RedBoard Artemis.

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

paulvha, 2021-04-26:
it can not find the header-file ( .h). remove the original sparkfun library and just use mine. All what the original library had is still in my library. Then use the examples as they are.

regards,
Paul

dyones7, 2021-04-26:
Ok thanks. I actually had 3 different libs, removed the two from Sparkfun and not it seems to compile fine.  Like I said just digging into Arduino fun.

Thanks again,
Dale

closed 2021-05-20 by paulvha