paulvha/ThingMagic #7: COMMAND_TIME_OUT seems to long

dyones7, 2021-06-16:
Hi Paul,

I have been using your updated SparkFun Simultaneous RFID Reader library with much success, specifically for reading to chips with deep user memory.

I issues however with the #define COMMAND_TIME_OUT set to 2000ms.

Have you modified this setting and do you know why it is set to 2 seconds. 

Are there commands that will have problems with this or could affect the operation of the Nano?

I wanted to get your opinion before I mucked with the library. 

thanks

paulvha, 2021-06-17:
it was like that. It is only a maximum timeout, in case the Nano does NOT respond a sae guard. If the Nano respond earlier it will react on that immediately. If it times out, you get an error message. The Nano is very sensative to power and will stop responding, this can help to prevent a dead-lock. Why 2 seconds ?? don't know, it is just how long you want to wait in case off no responds. The big Thingmagic / Jadak MercuryAPI is using 2 seconds as well.

regards
Paul

paulvha, 2023-01-26:
closing old issues

closed 2023-01-26 by paulvha