paulvha/ThingMagic #8: RESPONSE_IS_KEEPALIVE

attreewr, 2022-01-16:
Hi, on your example 1, Constant_Read, as compared to using the original library from Sparkfun, the code does not output the keep alive 'Scanning'.
    _if (responseType == RESPONSE_IS_KEEPALIVE)
        {
          Serial.println(F("Scanning"));
        }_
I have used the Keep alive as a point to look at when to get the TagReadtime off set from.
Thanks for the great addtions to the original library.

paulvha, 2022-01-17:
Thanks for your comment

I have checked the relevant part of the Example1 code but I do not see a difference. What do you experience ? Does the original exampe1 code work and show scanning and "my" example1 code does not ? Let me know I will try to find time to test later this week, but would be good to know the problem.
regards
Paul

attreewr, 2022-01-17:
HI, Yes exactly.
Original shows scanning, yours doesn’t.

if (responseType == RESPONSE_IS_KEEPALIVE)
    {
     // Serial.println(F("Scanning"));
      CurrMillis = now();  // **  Assign a time to the KEEPALIVE
…
}
Then when tag is read, getTagTimestamp add that value to the above to the exact REAL time was read.

I maybe using incorrectly but wanted to know when the Keepalive was triggered.

I need to assign an actual Time when the tag was read, the getTagTimestamp returns the time from the last KEEPALIVE, so if I don’t know when KEPALIVE happened, I can add that time to the getTagTimestamp

Thanks

closed 2022-01-17 by paulvha

attreewr, 2022-01-17:
> found the bug in the library and it is fixed in the new version just uploaded
> Thanks for pointing it out.

Thanks VERY much.