paulvha/ThingMagic #9: ThingMagic M5E

7ry4641n, 2022-04-12:
Hello. I am trying to connect ThingMagic M5E-C to ESP32 using your library but nothing works for me. I connected the M5E using a voltage divider on the TX line. I use stable power from the power supply. But I keep getting "Module failed to respond. Please check wiring" message. Could you help me? This is for my university project.

paulvha, 2022-04-12:
HI
Not sure this library is compatible with an M5E. Have not tried or studied that before.  BUT maybe...
How did you connect the hardware? (pictures ?)
you could turn on debug in the sketch and see what messages you get.

regards,
Paul
regards,
Paul

7ry4641n, 2022-04-12:
Getting with debug:
```
try to connect to RFID reader
sendCommand: 
 [FF] [00] [03] [1D] [0C]
Time out 1: No response from module
Try reset RFID speed
sendCommand: 
 [FF] [04] [06] [00] [00] [96] [00] [E0] [41]
Time out 1: No response from module
```
[Photos here](https://imgur.com/a/xGQO2pv)

7ry4641n, 2022-04-12:
Thanks a lot. [Here's Developer’s Guide for M5E](https://www.graftek.com.tr/dosya/m5efamilydevguide-may11-pdf/838/)
As I assume the problem is in different codes. I think to start the firmware I need to send the code 0xFF,0x00,0x04,0x1D,0x0B. I read it in the guide. Could you add M5E support to your library?

paulvha, 2022-04-12:
Thanks

I looked at the documentation and photos.

In the developers guide, page 65.. I see the same FF 00 03 1D 0C request for the M5E.  So maybe still a cable / connection issue
Where did you find 0xFF,0x00,0x04,0x1D,0x0B ??

Can not find information about the connection details to the M5E itself. Do you have that?

Looking at the pictures (hard to see) BUT is the 5K resistor connected to the same place as the 10K and yellow wire ??

regards,
Paul

7ry4641n, 2022-04-13:
> is the 5K resistor connected to the same place as the 10K and yellow wire ??
Yes that's right. Something is wrong?

paulvha, 2022-04-13:
could not clearly see from the picture, but that is good and correct.
Are you using serial2 ? which sketch do you use ? 
As mentioned earlier, I could not find info about the M5E connector. Have you double checked that it is connected to the right points ? What are the other connector point for ?

Regards,
Paul

paulvha, 2022-04-13:
With respect to the M5E connector, found that information and looks to be connected correctly.

Another test you can try: make a simple sketch to write something to the serial port and read from the serial port. Then connect RX and TX from the ESP32 together and check that you get the same data in return. 

Also I have validated in the source of the MercuryAPI that the command for requesting the version info is the same between the M6E and M5E.

regards,
Paul

7ry4641n, 2022-04-13:
> could not clearly see from the picture, but that is good and correct. Are you using serial2 ? which sketch do you use ? As mentioned earlier, I could not find info about the M5E connector. Have you double checked that it is connected to the right points ? What are the other connector point for ?

Sure, I'm using Serial2. ESP32 RFID Server sketch. 

[image]
[image]

> Maybe one more point. Do you have a USB to Serial convertor. if so connect the M5E to a PC and downlaod the Universal Reader Assistent (URA) from Jadak. Then you can test whether it works. Also it allows to create a communication log file to see what information is exchanged .

I can not download URA, the site is not responding. Can you send me the x64 version?

7ry4641n, 2022-04-13:
> Another test you can try: make a simple sketch to write something to the serial port and read from the serial port. Then connect RX and TX from the ESP32 together and check that you get the same data in return.

I'll try to test it now

paulvha, 2022-04-13:
drop me an email paulvha@hotmail.com and I will send you the URA..
Also see the other remarks that I have made in the meantime.

regards
Paul

7ry4641n, 2022-04-21:
Thanks to Paul, we figured out what was going on. So the problem was that, by default, M5e waits for a command at 9600 baud after power is applied, so it did not understand the command. Also, to start the reader, you need to additionally send the following command to it to start the firmware before configuring the device: `0xFF,0x00,0x04,0x1D,0x0B`
Hopefully @paulvha will someday add M5e support to the Sparkfun library to make things easier for newbies :)

closed 2022-04-21 by 7ry4641n