paulvha/ThingMagic #10: Compiling error when uploading to NodeMCU-32S

alex-prosh, 2023-02-04:
Hi, I am trying to use this library to connect a NodeMCU to an M6E Nano RFID Reader. I installed the library by downloading and extracting the folder to my libraries folder. The Arduino IDE seems to read the libraries and shows me the examples, but when I try running example1 constant read from this library I get an error:
`In file included from C:\Users\~~~~~\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp32/xtensa/xtruntime.h:30:0,
                 from C:\Users\~~~~~\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/portmacro.h:80,
                 from C:\Users\~~~~~\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/portable.h:96,
                 from C:\Users\~~~~~\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/freertos/freertos/FreeRTOS.h:105,
                 from C:\Users\~~~~~\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32/Arduino.h:32,
                 from C:\Users\~~~~~\AppData\Local\Temp\arduino-sketch-C473708062EFC059CA482FDCA5D4610D\sketch\Example1_Constant_Read_copy_20230204091135.ino.cpp:1:
C:\Users\~~~~~\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6/tools/sdk/include/esp32/xtensa/config/specreg.h:106:14: error: expected unqualified-id before numeric constant
 #define EPC  176
              ^
c:\Users\~~~~~\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library_dec2022\src/SparkFun_UHF_RFID_Reader.h:108:11: note: in expansion of macro 'EPC'
   uint8_t EPC[12];        // EPC bytes or part of the EPC to match
           ^

exit status 1

Compilation error: exit status 1`

**How do I fix this?**

paulvha, 2023-02-04:
I never tested on a NodeMCU but it looks to me that EPC is defined 2 times, once in the board library and another time in the RFID reader library. Looking at the version number of your boards library 1.0.6 that looks "old" to me. The version I have is 2.0.6 (same as I see online). Can you try to update and let me know ?

regards,
Paul

alex-prosh, 2023-02-04:
Hi Paul, 
For some reason it seems that the reference I was using contained only 1.06 as the latest version. Changing to to reference the official github with the link:
https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
Seems to have helped and I was able to install the latest version.

Unfortunately the errors are still present and seem the same but with the new version:

`In file included from C:\Users\prosh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/xtensa/include/xtensa/xtruntime.h:30,
                 from C:\Users\prosh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/portmacro.h:40,
                 from C:\Users\prosh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/freertos/include/freertos/portable.h:51,
                 from C:\Users\prosh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:63,
                 from C:\Users\prosh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6\cores\esp32/Arduino.h:33,
                 from C:\Users\prosh\AppData\Local\Temp\arduino-sketch-C473708062EFC059CA482FDCA5D4610D\sketch\Example1_Constant_Read_copy_20230204091135.ino.cpp:1:
C:\Users\prosh\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.6/tools/sdk/esp32/include/xtensa/esp32/include/xtensa/config/specreg.h:108:14: error: expected unqualified-id before numeric constant
 #define EPC  176
              ^~~
c:\Users\prosh\Documents\Arduino\libraries\SparkFun_Simultaneous_RFID_Tag_Reader_Library_dec2022\src/SparkFun_UHF_RFID_Reader.h:108:11: note: in expansion of macro 'EPC'
   uint8_t EPC[12];        // EPC bytes or part of the EPC to match
           ^~~

exit status 1

Compilation error: exit status 1`

alex-prosh, 2023-02-05:
I was able to solve this issue by using #UNDEF EPC directly before the EPC[12] section in SparkFun_UHF_RFID_Reader.h

closed 2023-02-05 by alex-prosh