paulvha/ThingMagic #11: Time out 1: No response from module when connecting NodeMCU-32S to EM6 Nano

alex-prosh, 2023-02-05:
When I try to connect the two boards through the Serial2 port of the NodeMCU I get the error:

Time out 1: No response from module
sendCommand: 
 [FF] [04] [06] [00] [00] [96] [00] [E0] [41]
Time out 1: No response from module
sendCommand: 
 [FF] [00] [03] [1D] [0C]
Time out 1: No response from module
Module failed to respond. Please check wiring.

How do I go about fixing this error / what should my next steps for debugging be?

paulvha, 2023-02-05:
I have never tested 2 different Nano boards connected to the same MCU. I wonder whether that would work. I can not test it as I do not have 2 boards. Also If you have the 2 working boards close to each their signals will interfere. 

But maybe you can share the sketch you are using and how you connect the wires. I can have a look and see whether I can give some advice.

regards,
Paul

alex-prosh, 2023-02-05:
I meant to say that I get this issue when I connect the two boards (NodeMCU-32S and M6E Nano) together through the UART interface that I get this error.

The sketch I am using is the example sketch, "Example1_Constant_Read" with the only change being DEBUG is set to 1 and I defined NanoSerial to Serial2, commenting out the SoftwareSerial portion.

My connections are as follows:
Node PIN16 -> M6E TX0 (Thru 15k pull down resistor)
Node PIN17 -> M6E RX1 (Thru 3.3V to 5V level shifter using VCC (+5V) and 3.3V of the Node MCU) 
Node GND -> M6E GND

M6E (-) -> PSU (-)
M6E (+) -> PSU (+)

paulvha, 2023-02-05:
Thanks. Now I better understand.

Is the PSU- connected to Node GND somehow ? 
What is the PSU voltage?
No need for the 15K pull down resistor. Remove it.
No Need for the level shifter on M6E RX. Remove it.

If anything (maybe not for test but for the long run). Make a voltage divider for the M6E TX. The output signal is between 0v and 5V which might be too high for the ESP32. So use 2 resistors serial : 5K6 and 10k . The middle is connected to the ESP pin 16, the other side of the 10K is to GND and the other side of the 5k6 is to M6E TX. 

regards,
Paul

alex-prosh, 2023-02-06:
Hi Paul,

It seems that the GND of the M6E Board and the (-) of the PSU are connected internally as the M6E's (-) is wired to ground. Just in case, I connected (-) on the PSU to the common GND on the NodeMCU and the M6E.

I removed both the 15k pull down resistor and the level shifter, however it still seems to give the same error message as though the board is not even connected.

I was using a voltage divider previously, but the resistance provided too much of a load on the with those values bringing the source TX voltage down from ~4.6 volts to ~3 volts (and the output from the voltage divider being aroudn ~2 volts) which is why I started using just the pull down resistor.

Either way, how do I fix this error?

Thanks,
Alex

paulvha, 2023-02-06:
I assume you have a 5V connected to the separate battery connector on the M6E. 
Is the power-led on?

I do not have an NodeMCU-3S, but looked at the datasheet (https://docs.ai-thinker.com/_media/esp32/docs/nodemcu-32s_product_specification.pdf).  I am a little confused.

I see PD16 is connected to GPIO16 and PD-17 is connected to GPIO17.  I also see that it shows U2RXD and U2TXD. 

However when I look in the driver (HardwareSerial.cpp) I see that GPIO16 and GPIO17 are used for the standard ESP32, BUT for an ESP32S3 U2RXD =19 and U2RXD= 20 (around line 77)

Did another device work on these Serial pins as Serial2.?
What happens if you connect pin16 and pin 17 together and try a loopback sketch?
Maybe try Serial1. According to the hardwareSerial.cpp that should be U1RXD P15 and U2TXT 16.

If the Serial is correct and working, do you have a USB/serial FTDI ? If so try to connect to a windowsPC and download the Universal reader Assistent (URA). At least you can then check the M6E is working. Else maybe you have another MCU board (Arduinol) and can try on that.

Regards,
Paul

alex-prosh, 2023-02-07:
Yes, I have 5V connected and the power led is on.

Perhaps the reason for this discrepancy is because the NodeMCU-3S is a different board from the NodeMCU-32S. If you were referring to the NodeMCU-32S then the driver seems to read: 
#elif CONFIG_IDF_TARGET_ESP32S2
#define TX1 17
indicating TX1 is mapped to GPIO 17 on line 64-65.

I've tried connecting the mcu to itself (16 <-> 17) and have tested Serial2.write("hello world!"); which has worked.

I haven't tried running on Serial1 yet.

Another device has not worked yet on the Serial2 of the NodeMCU pins, the Node boards are brand new.

I've tried running the URA and it has confirmed the functionality of the M6E although it took a few tries of reconnecting.

I can try running on the ESP8266 nodeMCU but this board cannot run the M6E nano without software serial if I am not mistaken which has been an issue as I was constantly getting the SRS Reset error.

paulvha, 2023-02-07:
I have indeed mixed them. (Too many ESP32 versions :-).
You have confirmed Serial2 is working and the M6E Nano is working with the URA. It is getting a challenge now as I am running out of ideas

From an earlier post, I see the code that has been sent to the M6E Nano:
the [FF] [00] [03] [1D] [0C] is requesting for the HW and firmware level. as it does not get a response it will try a different speed.
[FF] [04] [06] [00] [00] [96] [00] [E0] [41]
Then it tries again to get the HW and firmware level but still does not get an answer

I did have a couple of times that the M6E was not responding, but that was mainly because of the Power was not stable enough during start-up. it then went into an undefined state. I remove the power from the M6E and reconnect in order to reboot. 

A couple of questions : 
What happens if you connect 16 <->17 and run the M6e sketch with debug. 

I assume that you connect the wires to the M6E FTDI connection. Only direct wires between GND-GND, RX- TX (17) and TX-RX(16)  between. 

What happens if you do NOT use the external PSU, but connect also a wire to the FTDI VCC and the 5VIN on the NodeMCU. 

Also try to first power the board and load the sketch then connect the wire to the 5Vin and reset the board.

regards,
Paul

alex-prosh, 2023-02-10:
Hi Paul,

Thanks for your response, I was about to message you that it was again not working, but fortunately the last part of your message fixed it for me! I think the issue was with VIN not being 5V (it was 4.6 volts) connecting the +5V vcc of the level shifter to the psu which is exactly 5 volts and connecting it in the order you stated made it work perfectly although it was not scanning at first because the connection with the antenna was poorly connected.

Thanks,
Alex

closed 2023-02-10 by alex-prosh