paulvha/ThingMagic #12: Connecting ESP-32S nodeMCU yields incomplete response

alex-prosh, 2023-02-18:
When I try connecting to the M6E nano through a 5 V to 3.3 V bidirectional converter I get this error:
#1

sendCommand: 
 [FF] [00] [03] [1D] [0C]
response:  [DD] [84] [83] [7D] [72] [E3] [3E] [9E] [98] [0E] [23] [3E] [8C] [3C] [A4] [1C] [DC] [1F] [F6] [63] [FD] [18] [FF] [F0] [0F] [1C] [3F] [DE] [63] [FE] [7E] [7E] [7F] [FE] [FE] [E0] [FE] [EE] [EE] [A2] [A6] [A6] [7E] [80] [02] [E6] [40] [D0] [F0] [F0] [F0] [F0] [F0] [10] [FE] [FE] [EE] [A2] [02] [F8] [50] [00] [07] [1C] [F8] [FC] [3C] [0C] [CC] [CC] [CF] [CE] [CC] [CE] [0C] [E6] [EE] [EE] [E6] [82] [02] [50] [F0] [F8] [F8] [F8] [3C] [3C] [F7] [F8] [F8] [F8] [F8] [C6] [3C] [CC] [CE] [E7] [E6] [F7] [77] [76] [37] [76] [F7] [F8] [00] [02] [02] [50] [F0] [F8] [3C] [3C] [0C] [CC] [36] [3C] [3C] [3C] [3C] [F4] [1C] [CE] [F7] [37] [36] [B3] [37] [37] [37] [BB] [BB] [0E] [F8] [F8] [F0]
Corrupt response
sendCommand: 
 [FF] [04] [06] [00] [00] [96] [00] [E0] [41]
Time out 2: Incomplete response 9
sendCommand: 
 [FF] [00] [03] [1D] [0C]
response:  [2D] [56] [7A] [AD] [AA] [AE] [BF] [B7] [79] [5A] [6B] [A5] [0F] [0F] [0F] [0F] [0F] [C7] [F7] [C7] [0F] [0F] [0F] [0F] [0F] [CF] [C7] [0F] [0F] [0F] [C7] [0F] [FD] [79] [FD] [FD] [FD] [FD] [FF] [BF] [FF] [FF] [FF] [FF] [FF] [BF] [CF] [BF] [BF] [3F] [3F] [0F] [C7] [7F] [E0] [E0] [E0] [E0] [E0] [80] [80] [10] [00] [11] [59] [59] [59] [59] [FD] [07] [FD] [FD] [FD] [79] [0F] [FF] [2F] [0F] [0F] [0F] [CF] [0F] [0F] [7F] [E0] [7E] [7E] [7E] [7E] [7C] [7E]
Corrupt response
Module failed to respond. Please check wiring.

#2

sendCommand: 
 [FF] [00] [03] [1D] [0C]
response:  [E7] [B6] [DE] [7B] [9F] [FF] [CF] [E7] [0F] [00] [7F] [00] [86] [FA] [AF] [FB] [FC] [DF] [E6] [66] [66] [EE] [E6] [CA] [CE] [9C] [EF] [99] [AE] [7B] [79] [79] [DB] [A6] [7B] [7A] [7A] [5A] [F2] [B2] [66] [6E] [DE] [DE] [9E] [9E] [DE] [DE] [DE] [EE] [A6] [79] [7B] [7A] [6D] [7B] [79] [5A] [5A] [5B] [79] [5B] [DA] [92] [DB] [9B] [9B] [B6] [79] [7B] [7B] [7D] [D2] [5B] [9B] [B2] [66] [66] [66] [EE] [DE] [DE] [DE] [6E] [D2] [7B] [ED] [E9] [EB] [E9] [E9] [7B] [7B] [ED] [79] [7B] [7D] [DA] [DB] [92] [66] [5A] [79] [5A] [79] [9B] [5B] [59] [9B] [93] [B6] [66] [6E] [EE] [DE] [9E] [9E] [7E] [DB] [5B] [59] [5B] [79] [7B] [5A] [6D] [5B] [DB] [93] [9B] [93] [B2] [F2] [EE] [DE] [66] [66] [B6] [66] [DE] [66] [66] [DE] [DE] [DE] [9E] [9E] [3E] [7E] [FE] [E0] [9E] [DE] [EE] [6E] [6E] [6E] [6E] [6E] [66] [66] [66] [6E] [EE] [DE] [9E] [1E] [E0] [9E] [9E] [9E] [9E] [E1] [7E] [7E] [F8] [FE] [E0] [E0] [E0] [E0] [E0] [80] [80] [61] [9E] [9E]
Corrupt response
sendCommand: 
 [FF] [04] [06] [00] [00] [96] [00] [E0] [41]
Time out 2: Incomplete response 11
sendCommand: 
 [FF] [00] [03] [1D] [0C]
response:  [8A] [9F] [66] [B3] [7D] [BF] [BF] [CB] [5B] [DB] [DB] [C9] [4B] [CC] [B3] [7B] [6C] [B5] [2C] [54] [6B] [6F] [CB] [DB] [EF] [3A] [C7] [CF] [0F] [0F] [C7] [0F] [CF] [0F] [0F] [0F] [0F] [CF] [3F] [3F] [3F] [FF] [FF] [FD] [FD] [FD] [FF] [FD] [FD] [FF] [FD] [FD] [FF] [FD] [FF] [CF] [FF] [FD] [FD] [FD] [AF] [FD] [BF] [0F] [C7] [FF] [7D] [11] [01] [00] [80] [00] [11] [59] [00] [01] [11] [59] [11] [FD] [FD] [FF] [FD] [7D] [19] [F8] [79] [FD] [FF] [3F] [0F] [FD] [7D] [59] [F9] [59] [19] [10] [58] [FD] [FD] [FD] [FF] [FD] [FD] [BF] [F7] [BF] [FF] [FF] [FF] [FD] [3F] [2F] [0F] [0F] [C7] [FF] [FF] [FF] [FD] [3D] [FD] [FD] [79] [F9] [59] [FD] [FF] [FF] [FF] [3F] [BF] [FF] [FF] [FF] [3F] [0F] [0F] [0F] [0F] [C7] [FD] [FD] [FD] [FD] [FF] [FD] [FD] [FD] [59] [59] [FD] [FD] [FD] [FF] [FF] [FF] [FF] [FF] [FD] [FF] [BF] [3F]
Corrupt response
Module failed to respond. Please check wiring.

alex-prosh, 2023-02-18:
it works when the connection the level shifter is not used and the connection is direct from the MCU to the M6E, but could this cause issues in the long run if I power the 3.3 V NodeMCU RX port with 5 V from the M6E?

paulvha, 2023-02-19:
The response is indeed incorrect. I had the same issue with ESP32. The root cause turned out to be that the level shifter on the M6Nano board (TXB0104) does not work well in conjunction with another level shifter. Hence in my implementation, I used a resistor divider on the ME6-TX line.  You can take nearly any value (the current is very low) as long as it is 3 : 2 division. So 10K / 22K should work as well.

```
TX- nano  --- 1K ---!---2K--- GND
                    !
             RX ESP32
```

regards,
Paul

closed 2023-03-25 by alex-prosh