paulvha/ThingMagic #13: esp32 continuous reconnection to m6e nano

alex-prosh, 2023-03-25:
Hi Paul,

I am using your library with the #undef statement included to connect from the esp32 to the m6e nano. However, I see that in the constant read example code we are simply stepping into an infinite "while" loop if we are unable to initially connect to the M6E Nano. My question is: how can we make it so that the esp32 continuously attempts to reconnect to the M6E Nano rather than hanging when it is unable to do so the first time?

When I try to do this myself by removing the while(1) loop, replacing it with while(!setupNano(115200)) and printing the nano.msg I keep getting the code:
01 00 03 1D 0C C2 00 A4 60 30 00 00 02 20 19 06 10 01 19 01 11 00 00 00 10 BC 91

Also, does the scanning of tags function asynchronously? I am trying to scan a tag and send its information to a web server and I am wondering if there could be periods during the sending process that the scanner is "inactive" or scanning, but the esp32 is not able to read it because it is currently handling an HTTP POST request.

Thanks

Alex

paulvha, 2023-03-25:
HI Alex

In the latest version (March 2023) I have handled the EPC conflict with renaming.

For sure you can change the loop to constant check, but I wonder whether that helps. In the SetupNano() there are already 2 attempts (one without doing a speed change and one WITH doing a speed change). If that fails I normally find there is a problem with Hardware: wiring or the M6e had a power dip and is in limbo.
 
Not sure how you connect the ESP32 and M6e, but also the 115200 could be a reason for failure. The data you see in msg is a mix. Normally you should see a sequence like this: 

ESP32 Sending:   FF  00  03  1D  0C
ESP32 Receive: ff 14 04 00 00 14 12 08 00 30 00 00 02 20 19 09 23 01 09 02 02 00 00 00 10 6c 33

For sending the data: Once you detect a tag that you want to send, just call a function to send that data. Once completed sending it returns to the loop() and will continue and look again at whether a tag was detected and is stored in the M6e buffer. The M6E buffer can store a maximum of 48 96-bit EPC tags.  

Regards
Paul

alex-prosh, 2023-03-25:
Hi Paul,

Thanks for your quick response. What baud rate would you recommend? 38400 or 9600? Also, the reason why I want to implement a constant reconnect loop is that sometimes my m6e nano gets stuck in a possibly undefined state for which the solution is to manually reset the esp32 a few times. If I were to implement this resetting automatically I think it would help with this issue.

paulvha, 2023-03-25:
38400 is a good baud rate that works most of the time.

The issue about the M6E getting stuck I have observed a number of times. Think "M6e had a power dip and is in limbo". I connected scope once to see what happens and I see constant spikes with a HIGH frequency. The only way to recover was remove the power and reconnect.  

Alternative to removing the power is the enable line (e.g. connect that to a GPIO on the ESP32):  
"The ENABLE line (referred to as the SHUTDOWN line in the M6e) must be pulled HIGH or left unconnected in order for the module to be operational. To shut down the module, the line is set LOW or pulled to Ground. Switching from high to low to high is equivalent to performing a power cycle of the module. All internal components of the module are powered down when ENABLE is set LOW." (source :  M6e design document)
regards
Paul

alex-prosh, 2023-03-29:
Hi Paul,

I've tried removing the power source and reconnecting which works sometimes, but many times it will still show the same error message with the same code. Are there any diagnostic tests I should perform? And, are there any surefire steps I should take to make sure all the states are reset and that the esp32 is always able to connect immediately to the m6e nano?

Thanks,

Alex

alex-prosh, 2023-04-05:
Hi Paul,

Here is an image illustrating the connections, let me know if anything is confusing as there are a lot of wires...

[image]

The red wire with the black slash through it is not connected to anything. Also, do you know what the difference is between the VCC on the serial pins and the actual "+" connection for external power supplies? Since I am not using a serial decoder like the ones used to connect this to a computer through a USB, can I use the VCC serial pin as my +5V Input? When I try doing so the power led turns off.

The power source I am using is this: [link to amazon](https://www.amazon.com/NOYITO-Precision-100V-264V-Step-Down-Ultra-Small/dp/B07FMZ3Z2K/ref=sr_1_5?crid=3L43QNF5MOE4C&keywords=120v%2Bac%2Bto%2B5v%2Bdc&qid=1678403955&s=electronics&sprefix=120v%2Bac%2Bto%2B5v%2Bdc%2Celectronics%2C153&sr=1-5&th=1)

The setting for read power is 5 dB

This happens when I try to connect.

The changes that were made to the sketch were commenting the software serial portion and changing the define to Serial2.

Thanks,

Alex

paulvha, 2023-04-06:
HI

The ' +' on the M6E Nano connects through a diode to the VCC on the Nano. If you have not broken JP1 on the nano it will also feed the VCC pin on the serial connector.

One aspect I do not understand with this wiring is where the ESp32 is getting it's power from? I see the GND connection, but how is it getting '+' ?

regards,
paul

[IMG-1502]<https://user-images.githubusercontent.com/56415559/230225060-e6683192-3a54-4476-a404-2583715951d9.jpg>

alex-prosh, 2023-04-06:
Hi Paul,

I checked the voltage on the VCC while the "+" is connected to 5V and it seems like it is 0 V. Does that mean the JP1 connection is fried?

The ESP32 is getting its power from the mirco-usb connection. But, here is an updated system I made that does not use the micro-usb and instead gets its power directly from the PSU. I've also added a few lines of code to turn on the LED in the setup and to turn it off if it gets past the setupNano() function in the setup (which it has yet to do).

[image]

alex-prosh, 2023-04-07:
Never mind about the serial connection. I desoldered the jumper allowing for the M6E to be powered through the serial connection.

alex-prosh, 2023-04-07:
On the Sparkfun hookup guide for the M6E nano they state "Note: The FTDI Cable products are known to be incompatible with the Simultaneous RFID Reader. Please use the CH340 Serial Basic Breakout or other serial interface." Do you think perhaps wiring the ESP32 through the actual RX/TX pins present on the M6E nano might work better? I could also try powering the M6E nano with 5V directly connected to the 5V pin on the board meant for the arduino uno as Sparkfun themselves state that this method of Target Power is best.

paulvha, 2023-04-08:
The issue with the FTDI cable is because of the resistor that is serial to the RX line. Spend a good time on that : https://github.com/sparkfun/Simultaneous_RFID_Tag_Reader/issues/10

Not sure what you mean with "the actual RX/TX pins present on the M6E nano". If these are the software serial pins, I would say NO. The pins RX/TX (hardware) and D2/D3 (soft) are actually connected to the switch HW/SW, The center to that switch is connected to FTDI RX/TX and the M6E Nano.

I think a direct 5V to the board is better, but make sure this supply has a good capacitor. The nano works in burst and a normal wall-wart will not be enough.

What is on this moment still the problem ?
Instead of a 5V powersupply connected to the board : try using a charged lipo-battery. 3v7 will do for test

regards,
Paul

paulvha, 2023-04-08:
Thanks Alex

Now I see the 5V is connected to the the ESP32.

Is it working or not ?
If not working what is not working?
Do you get error message?
What does the sketch looks like ?

regards,
Paul

[unnamed]<https://user-images.githubusercontent.com/56415559/230460281-d170d620-d8e4-4fb7-9fd0-0d1f1808a016.jpg>

[unnamed]<https://user-images.githubusercontent.com/56415559/230460281-d170d620-d8e4-4fb7-9fd0-0d1f1808a016.jpg>

alex-prosh, 2023-04-10:
Hi Paul,

Thank you for your responses. Currently this has been the most effective solution and has not failed me once. You are saying that if I wire to the jp1 connection's rx/tx it would be better? As you can see I've currently wired it to the hardware rx and tx.

[image]

In terms of the code, I found that reducing the baud rate of setup nano to 38400 rather than 115200 has been effective. 

Also, there have been very strange grounding issues. Perhaps it is due to the quality of the wires and breadboard, but I find that sometimes it will not work if I do not add a redundant ground connection to some of the components.

I have removed the antenna and desoldered some of the jumpers for simplicity sake, but I plan on adding them soon enough. Do you think my current set up would have any issues?

paulvha, 2023-04-10:
level converter on the ESP32 TX to M6E nano RX is normally not necessary. The 3V3 output worked for me all the time.
I am concerned that your capacitors on the power supply might not be enough. I would add a larger capactor  e.g. 1000uF
You can connect the TX and RX wires at any place. Personally, I always connect to the TX/RX FTDI connection. There have been boards in the past where the UART switch was not optimal.

38400 speed is enough. works stable. 

The grounding issue can indeed be the result of the wiring and how they connect to the board.

The removal of the power /setting enable pin low, will cause a complete reset of the M6E. It needs time to stabilize. maybe a longer delay after enabling power again. but it could also be a spike in the power supply. The M6E is sensitive to power.

regards,
Paul

microElabDevelop, 2023-10-03:
@alex-prosh hi hopefully your fine. can you  guide me to perfome basic funcation of m6e nano(basiclly i use  M7e-Pico) with esp32 read write etc. if give me arduino code its very helpfull.

paulvha, 2023-10-03:
Hi

Interesting... not worked with the M7E before.

You can download the software with zip-file for the ArduinoIDE from https://github.com/paulvha/ThingMagic/tree/master/Arduino_lib_special.

Use Serial1 pin 16 /17

comment out :

// if you plan to use softserial:  initialize that here, else comment it out the 2 lines
//#include <SoftwareSerial.h> //Used for transmitting to the device
//SoftwareSerial softSerial(2, 3); //RX, TX

Select Serial1
// define the serial port to use (E.g. softSerial, Serial1 etc)
#define NanoSerial Serial1

IF THAT DOES NOT WORK TRY :

Hardware connect

M6E Nano     ESP32
VCC --------    VUSB
RX  --------     TX  pin 26 !! // see below
TX  --------     RX  pin 25 !! // see below
GND --------   GND

Software connect
in Setup() :
BEFORE : if (setupNano(38400) == false) //Configure nano to run at 38400bps

 //Initialize communication port
 //FOR ESP32 SERIAL1 CHANGE THIS TO : E.G. RX-PIN 25, TX-PIN 26
 NanoSerial.begin(38400, SERIAL_8N1, 25, 26);

regards
Paul