paulvha/ThingMagic #16: Need SELECT command example

dyones7, 2023-11-14:
Hi paulhva,

Can you add a select command so that the sparkfun reader fix the problem in this link:

https://github.com/sparkfun/SparkFun_Simultaneous_RFID_Tag_Reader_Library/issues/12

I am trying to read the Axzon sticky tags temperature and I don't know enough to move Jadak's  suggested code to the library. I have also asked sparkfun if they can help as well.

Thanks you for what you do.

paulvha, 2023-11-15:
Adding GEN2 code is not a straightforward task. It would mean creating a debug trace with the Jadak library and then translating that into working functions on the Sparkfun library and then it needs to be tested.  a lot of time.

I tried to obtain the PDF document of AXZON-AFR tag, but I am hesitant as It asks me to put in my credentials to get access. I helped some people in the past and some tags just work by reading from a specific offset address:(either user / TID or  EPC memory. For that reason, I have created examples like 18, 19 and 20.

As for help from Sparkfun, I have not seen an update of the original Sparkfun library for years. I don't expect they will make any update.

regards,
paul

dyones7, 2023-11-15:
> WIKA notification: This is an external email. DO NOT reply, click links or attachments unless you know the sender

Hello paulhva,

Thanks your reply and for clarifying what this task would entail.

I have read other sensor type tags and it is true that most require only a "READY" to access the data - some older ones required special firmware on the reader of choice. This tags is different in that the require the SELCT command be given to enable the A2D conversion process and also requires the UHF filed be left on for a further 2-3 ms.

I have contacted the manufacture of the IC to see if they can tell me how this works or to explain if there is another way to trigger the conversion.

After more investigation, I thought the SELECT command requires a full or partial EPC to select a group or single tag - similar to your example 20 ( and the reason for the request).
However, in the data sheet (which I'm not sure I can share) it appears that they are addressing a specific word in USER memory with the SELECT command. To me I don't see how this has anything to do with "selecting" a tag per GEN2 standards, but maybe it has other functionality I don't know about.

I will update any status that I find if there is any other interest.

Thanks again for your help and insight.

paulvha, 2023-11-16:
I think what you describe (access directly a USER or specific EPC address is what i have seen before. Some of those addresses (especially in TID and EPC) are outside the normal (maximum) address space. I tried to access this (my) morning again and got access to the product brief without the need to provide information. I see that the chip used is the Magnus chip. On Internet I found a description of that and it looks to me you can just read the temperature with standard read commands.  Have you tried one of the examples and see whether it works.

dyones7, 2023-11-16:
Hello paulhva,

The manufacture of the Magnus chip sent me this:

"Apps use the SELECT command mostly to singulate part of the EPC or the full EPC. But SELECT commands can be used in other memory banks. For example, many apps do SELECT commands in the TID to singulate tags of a specific manufacturer and/or specific model.

Magnus-S3 has two word locations in the USER bank (word addresses: 0x0D and 0x0E)  that can be used to activate the On-Chip RSSI measurement and the Temperature measurement...."

In the spec I have it says that you must issue the SELECT as shown, then perform the READ to get the data from the chip.  I am working with the manufacture to get more definition the command structure.

Thanks,

drbadarmuneer, 2024-06-28:
Hi Dear,
Did you manage to get an example of SELECT COMMAND to read the temperature code on Magnus S3? I would appreciate if you could share the example, I am stuck with this one.

dyones7, 2024-06-28:
Hi paulvha,

It has been a while since I worked on that tag project but here is what I can recall.

I think we needed to use the ThingMagic Sargus RFID reader as not all readers will execute the Select Command as required.

I was provided example code from Axzon support below. See the response from them below. I think you can download from the github link below.

Good Luck,
dy

Hi Dale,

- You can buy a handheld CS108 reader: https://www.convergence.com.hk/cs108/ . They have apps for Android and iOS. It reads the temperature very fast and it can easily simultaneously read many tags in front of the reader

- You can borrow from us an old Nordic ID Merlin. It is slow and it can only handle one tag in front of the reader. With multiple tags you don't have control of which one it will read. But it provides accurate temperature measurements

- With the old ThingMagic readers or Nordic ID AR62 or AR82 you probably have to update the firmware. Then download the following source code from GitHub: https://github.com/Axzon/reader-sample-code. And use Microsoft Visual Studio to compile the code. These are very simple examples that work in terminal mode

- The Zebra FX9600 is capable of reading the temperatures. But it is the same as the ThingMagic. I can give you a very simple source code example that you would have to compile.

- Impinj readers can not read the Magnus temperature sensor

paulvha, 2024-06-30:
I have not done any further work on trying to read the Magnus tags.

regards,
Pau

drbadarmuneer, 2024-06-30:
Dear PaulThank for your reply.Regards Badar